afry-south/dragonfly-fcb #15: Migrate code to STM32CubeF3 library

DanielStenbergAF, 2015-04-20:
Current code is written for the old Std Peripheral Library, which is no longer maintained or recommended for new applications. 

Instead, the STM32Cube library is recommended.

Get acquainted with it and migrate old code to use the new library functions instead.

Perhaps even make use of the STM32CubeMX windows app, which help developers calculate clocks, prevent I/O conflicts and even does some auto-code generation of peripheral initialiation.

DanielStenbergAF, 2015-05-08:
Commented out old code and made the program build and run on the board.

Gradually, the old Std Periph Lib code will be migrated (and improved).

Most of the motor control code has already been migrated.

DanielStenis, 2015-05-26:
Motor control has been done.

Receiver code in progress (quite far along).

What remains?
Sensor reading
Control allocation

DanielStenis, 2015-07-28:
Done. (Older code will be reused/replaced, commented out for now)

closed 2015-07-28 by DanielStenis