afry-south/dragonfly-fcb #18: Kalman estimation: Approximate sensor noise variance and offset

DanielStenbergAF, 2015-04-20:
To improve the Kalman filter algorithms for estimation of:

- Roll
- Pitch
- Yaw rate
- Upward speed/acceleration

the variances and offsets of the sensors play a great part. Read up on how they affect the linear Kalman algorithm.

Either calculate the values or look in datasheet of the sensors.

DanielStenis, 2015-09-08:
Two very similar suggestions on fusing gyroscope and accelerometer (we may use magnetometer instead for correction?) in a Kalman filter:

http://www.linushelgesson.se/2012/04/pitch-and-roll-estimating-kalman-filter-for-stabilizing-quadrocopters/

http://se.mouser.com/applications/sensor_solutions_mems/

Perhaps a suitable algorithm to start with before more advanced sensor fusion and EKF.

DanielStenis, 2015-09-09:
Use magnetometer to get attitude corrections?
Roll/Pitch estimation: Gyroscope and Magnetometer/Accelerometer
Yaw rate estimation: Gyroscope and Magnetometer
Vertical velocity estimation: Accelerometer and barometer(altimeter)