afry-south/dragonfly-fcb #2: Improve magnetometer calibration

DanielStenbergAF, 2015-04-20:
Improve magnetometer calibration using Gauss-Newton method

Read more:
https://chionophilous.wordpress.com/2012/09/01/implementing-the-gauss-newton-algorithm-for-sphere-fitting-1-of-3/

adam-at-epsilon, 2015-08-12:
Full scale is set to 1.3 Gauss as earth's magnetic field vector is .5 Gauss according to Wikipedia.

DanielStenis, 2015-08-12:
Very good, should give us more than enough wiggle-room for any magnetic interference.

There are some datasheet and application notes from ST on calibrating the magnetometer.

I previously implemented an algorithm that would compensate for hard-iron and soft-iron distortion of magnetometer readings. You basically need to turn the quadcopter/fcb board 360 degrees around all of its 3 axes to calibrate it, but will likely only need to do this once.

Anyway, the link in the OP suggests an improvement on what I did and also provides a fun mathematical optimization method!

DanielStenis, 2015-09-08:
Hard-iron offset it pure scaling of x, y, z measurement axes.

Soft-iron offset has distorted/skewed the magnetometer measurement sphere.

We will start with the hard-iron offset. If we do not have any strong magnetic sources on-board, this will most likely be sufficient to gather good measurements.

DanielStenis, 2015-09-08:
Also, we may not need to use float values (Gauss values). Raw values enough if we want to calc attitude with magnetometer?

DanielStenis, 2015-09-11:
* Use dedicated sampling task for this? 
* Store sampled values dynamically?

adam-at-epsilon, 2015-09-14:
General advice for doing numerical solutions (from link in 1st post)

_Choosing parameters that are about the same order of magnitude leads to stable, reliable performance._ because we avoid over/underflow of floating-point values.

adam-at-epsilon, 2015-09-20:
consider offline calibration by reading the values to a file, then doing the Gauss-Newton Least Sphere Fit in Scilab, GNU Octave or similar.

Pro:
Could be useful if we are running out of program

Con:
FCB must be attached to a PC during calibration, this may not be practical.

adam-at-epsilon, 2015-10-04:
The magnetometer is now compensated by the Gauss-Newton (offline, implemented in SciLab) method with respect to offset & scaling.

adam-at-epsilon, 2015-10-04:
@DanielStenis 

are any scripts using the human-readable variant of the `start-sensor-sampling` on the Termite console? If not, I would like to remove the line breaks for easier text processing for the offline variants of the Gauss-Newton calibrations.

DanielStenis, 2015-10-04:
Not at the moment, no. But you could implement a new output encoding type named commaseparated_encoding or something. If we want to keep the human-readable as it is for now

DanielStenis, 2015-10-08:
Some more info about hard/soft-iron compensation can be found here:

http://www.sensorsmag.com/sensors/motion-velocity-displacement/compensating-tilt-hard-iron-and-soft-iron-effects-6475

adam-at-epsilon, 2015-10-08:
So I implemented an extra "c" parameter for sensor - sampling. All it does is to print the values for XYZ on one single line.

DanielStenis, 2016-05-09:
Should we care about magnetometer declination (correction for difference between magnetic and geographic north pole)?

closed 2016-05-16 by superdupersvard