afry-south/dragonfly-fcb #3: Connect and read barometer/altimeter

DanielStenbergAF, 2015-04-20:
Connect the barometer to FCB and read input. Recalculate input to give a rough estimate of altitude.

adam-at-epsilon, 2015-10-09:
@DanielStenis do we have a datasheet for the barometer?

DanielStenis, 2015-10-15:
I'm sure one exists somewhere. Check the breakout board on the quadcopter for a model number.

adam-at-epsilon, 2015-10-16:
Ah yes, the barometer is already attached to the 'copter. This is the board

https://www.sparkfun.com/products/11824

Here's a link to the datasheet:

http://cdn.sparkfun.com/datasheets/Sensors/Pressure/BMP180.pdf

DanielStenis, 2015-10-19:
Exactly, but we need to solder on a pin header to it and connect this to the FCB I2C pins.

For this to be possible, we need to update the FCB breadboard layout so we get a nice way to connect external sensor to this bus. Right now, it is not clear how this should be done.

Maybe we could bring in Abed to have a look at connecting external devices in a more convenient way? He could even dig up the current solution for the receiver and motors and suggest improvements.

adam-at-epsilon, 2015-11-06:
re-open Issue #17 when this gets some traction.

adam-at-epsilon, 2015-11-11:
There's some driver code available from BoschSensortech on GitHub that we can use:
https://github.com/BoschSensortec/BMP180_driver

superdupersvard, 2016-06-23:
Driver code for BMP180 are now merged back to master.

closed 2016-06-23 by superdupersvard