afry-south/dragonfly-fcb #7: Improve sensor reading using interrupt and DMA

DanielStenbergAF, 2015-04-20:
Improve the sensor data reading from accelerometer, magnetometer and gyroscope.

By using an interrupt for when sensor data is ready (DRDY pin/INT1/IN2 pins from sensors) and reading it through DMA - much more efficient sensor reading and CPU usage.

Read more: http://letanphuc.net/2014/06/stm32-mpu6050-dma-i2c/

adam-at-epsilon, 2015-07-29:
Using the INT1 pin to send an interrupt to CPU now works, but we really want one of the acc or mag interrupts to go through the DRDY pin on the LSM303DLHC.

adam-at-epsilon, 2015-07-29:
The pulse width of the INT1 set to DRDY1 using the oscilloscope is 750microseconds.

adam-at-epsilon, 2015-07-29:
According to chapter "I2C Operation" in LSM303DLHC" Data Sheet, multiple read/write seem possible though the current STM32Cube code does not appear to take advantage of this.

adam-at-epsilon, 2015-07-30:
the actual time to read the accelerometer is 480 microseconds

adam-at-epsilon, 2015-09-04:
Interrupts now work for Gyroscope, Accelerometer and Magnetometer.

Still work to do according to previous comment and also with DMA:

http://letanphuc.net/2014/06/how-to-use-stm32-dma/

http://letanphuc.net/2014/06/stm32-mpu6050-dma-i2c/

adam-at-epsilon, 2015-09-08:
yet another link

http://www.embedds.com/using-direct-memory-access-dma-in-stm23-projects/