afry-south/dragonfly-fcb #20: Change IDE to Eclipse or other?

DanielStenbergAF, 2015-04-29:
CooCox is alright, but doesn't seem to be the most stable IDE and has rather messy configuration.

Perhaps consider another open-source IDE? Can we use Eclipse with GCC for ARM Toolchain with debugging and programming directly from the IDE?

adam-at-epsilon, 2015-05-07:
I concur, if we can find a more stable and full-featured IDE in the future like eclipse it would be good.

DanielStenbergAF, 2015-05-07:
GNU ARM Eclipse Plug-ins: http://sourceforge.net/projects/gnuarmeclipse/

These plug-ins provide build and debug extensions for Eclipse CDT (C/C++ Development Tools) for 32/64-bit GNU ARM toolchains like GNU Tools for Embedded, Linaro, etc, ready to run STM32Fx project templates and full integration for advanced J-Link JTAG/SWD probes, including SWO tracing console and peripheral registers view.

DanielStenis, 2015-05-31:
I think we should consider getting GCC for ARM running with Eclipse soon as it would open up the possibility to work with the project in both Windows AND Linux OS.

https://launchpad.net/gcc-arm-embedded has a GCC ARM version for Linux

According to moderators on CooCox CoIDE forum, a Linux version of their IDE is currently not under development.

DanielStenis, 2015-05-31:
More IDE options here: http://www.emcu.it/STM32/What_should_I_use_to_develop_on_STM32/What_should_I_use_to_develop_on_STM32.html

DanielStenis, 2015-06-01:
For Linux, we must also consider ST-Link/V2 debugger support and drivers. So in summary we need:
- Compiler/linker/toolchain: GCC for ARM (has Win/Linux/Mac versions)
- IDE: Eclipse?
- Debugger drivers: Drivers exist for Windows. What are out alternatives for Linux?

adam-at-epsilon, 2015-06-16:
Eclipse + OpenOCD + Windows

http://engineering-diy.blogspot.se/2012/12/stm32f3-discovery-on-windows-with.html

DanielStenis, 2015-07-28:
http://gnuarmeclipse.livius.net/

adam-at-epsilon, 2015-08-03:
Eclipse has an Egit plugin for git. This could be useful when refactoring our source tree.

http://www.eclipse.org/egit/

DanielStenis, 2015-08-03:
Got GNU ARM Eclipse plugin to work for Eclipse CDT. Shouls be a possibility for both Windows and Linux OS.

Also got OpenOCD debugging working with the ST-Link in Eclipse.

See eclipse-migration-branch branch in repo.

DanielStenis, 2015-08-09:
Using EGit at my home workstation. Works well it seems.

DanielStenis, 2015-08-12:
Done.

closed 2015-08-12 by DanielStenis