afry-south/dragonfly-fcb #31: use FreeRTOS for handling USB

adam-at-epsilon, 2015-05-07:
It would be desirable if USB was handled in a task running at the lowest priority.

adam-at-epsilon, 2015-05-07:
The below is an approximate call stack for USB handling.

```
./fcb/src/stm32f3xx_it.c:232:  HAL_PCD_IRQHandler(&hpcd);
|
V
stm32f3xx_hal_pcd.c:553:    PCD_EP_ISR_Handler(hpcd);
|
V
static HAL_StatusTypeDef PCD_EP_ISR_Handler(PCD_HandleTypeDef *hpcd)
|
V
void HAL_PCD_DataOutStageCallback(PCD_HandleTypeDef *hpcd, uint8_t epnum)
|
V
USBD_StatusTypeDef USBD_LL_DataOutStage(USBD_HandleTypeDef *pdev , uint8_t epnum, uint8_t *pdata)
|
V
./Core/Src/usbd_core.c:331:    pdev->pClass->DataOut(pdev, epnum);
|
V
USBD_CDC_DataOut # this is a callback in the USBD_ClassTypeDef structure
|
V
*hcdc pdev->pClassData
((USBD_CDC_ItfTypeDef *)pdev->pUserData)->Receive(hcdc->RxBuffer, &hcdc->RxLength);

```

DanielStenis, 2015-05-20:
Here is some information about handling USB with FreeRTOS on STM32:
http://www.pezzino.ch/stm32-and-freertos-usb-communication/

adam-at-epsilon, 2015-05-20:
a crude measurement was done by sending 32byte messages over the USB port and then piping the output to a file via Putty which resulted in a current speed of about 64Kb/s.

Windows OS says the difference between the "created" and "last modified" date was 38 seconds and the file contained 2456116 bytes.

DanielStenis, 2015-07-19:
This is more or less done. Looking at using the FreeRTOS+CLI (Command line interface) to organize command and data transmission.

DanielStenis, 2015-07-28:
DOne.

closed 2015-07-28 by DanielStenis