afry-south/dragonfly-fcb #33: trim task call stack size

adam-at-epsilon, 2015-05-07:
Task call stack size is supplied as a parameter whenever a task is created.

Giving a too large stack uses up memory unnecessarily. A too small stack size will result in memory being overwritten elsewhere.

DanielStenis, 2015-08-31:
Perhaps this could even be trimmed for each task individually?

adam-at-epsilon, 2015-09-27:
define INCLUDE_uxTaskGetStackHighWaterMark

This will enable this: http://www.freertos.org/uxTaskGetStackHighWaterMark.html

Which allows us to see how much of the task stack has been used.

adam-at-epsilon, 2015-09-28:
`vTaskGetRunTimeStats` uses `sprintf`, not `snprintf`

adam-at-epsilon, 2015-10-07:
with the latest check in task-status prints lowest historical watermarks to console.

Use this to trim stack sizes.

adam-at-epsilon, 2015-10-07:
our current stack sizes are, at idle (rightmost numbers, alignment doesn't seem to work ...):

The number meas the lowest number of bytes ever available in the stack so far.

    Task			 Abs Time	 % Time 	Stack (rem b)
    USB_COM_RX       50215       14%	        313
    IDLE                      224294     62%         104
    FLIGHT_CTRL        66043       18%         88
    Tmr Svc	           0              <1%	        220
    tFcbSensors       17677       4%           391
    USB_COM_TX        1              <1%          92