afry-south/dragonfly-fcb #34: Unit testing setup

DanielStenbergAF, 2015-05-07:
Is there a way where we can test some (all?) of the code functions through unit testing?

What would it require? Is there some standard method of unit testing embedded applications such as the one run on the STM32F3Discovery board?

Do we need to copy-paste function from target code to a separate C environment?

DanielStenis, 2015-08-12:
Currently working on system emulation with QEMU, which opens up possibilities to unit test greatly.

DanielStenis, 2015-08-12:
How does unit test on real HW work?

adam-at-epsilon, 2015-09-08:
I think we should add a stability test on hardware as well, perhaps if we could use a power source other than the battery (or two batteries), we could perhaps run the software for 2-3 hours in a row or preferably even longer.