afry-south/dragonfly-fcb #36: fcb-fsm protocol

adam-at-epsilon, 2015-05-07:
It is suggested that we use [https://developers.google.com/protocol-buffers/](Protocol Buffers) or something similar to serialise and deserialise data for communication between FSM and FCB.

It has to be lightweight in order to not overload FCB which has real-time constraints.

DanielStenbergAF, 2015-05-07:
http://koti.kapsi.fi/jpa/nanopb/

This is the best implementation I've found for embedded-C. It is already suggested for fcb<->PC communication.

DanielStenis, 2016-04-15:
Suggested protocol over RS232/UART communication integrates with the CLI:

>>> get-sensors p\r
<<< [1 byte msg ID][2 byte msg len][protobuf data of specified len]\r\n

DanielStenis, 2016-05-08:
Added protobuf header with CRC to some responses:
get-sensors p\r
get-states p\r
get-motors p\r
get-receiver p\r

Development in branch: 36-usbuart-protocol

DanielStenis, 2016-06-06:
get-ref-signals p\r
get-ctrl-signals p\r

Have been done. Pending verification before 36-usbuart-protocol merged to master.