afry-south/dragonfly-fcb #40: Increase USB communication speed

DanielStenbergAF, 2015-05-11:
The USB is configured to emulate a UART RS232 com port, currently set to communicate at 115200 baudrate. Investigate if this can be configured to higher speed since USB allows much higher data rate. 

This may be needed to ensure sensor data samples etc can be transmitted with high enough sample rate.

adam-at-epsilon, 2015-05-21:
it seems that using snprintf, we can only use 64 bytes at a time and longer messages get fragmented.

This is by trial-and-error, the cause could either be the STM32 or Windows USB driver.

DanielStenis, 2015-05-21:
https://my.st.com/public/STe2ecommunities/mcu/Lists/STM32Discovery/Flat.aspx?RootFolder=https%3a%2f%2fmy%2est%2ecom%2fpublic%2fSTe2ecommunities%2fmcu%2fLists%2fSTM32Discovery%2fUSB%20HID%20or%20%20CDC&FolderCTID=0x01200200770978C69A1141439FE559EB459D75800084C20D8867EAD444A5987D47BE638E0F&currentviews=1618

Discussion regarding USB Speed, is 1.2 MB/s possible?