afry-south/dragonfly-fcb #42: LaTeX documentation

DanielStenbergAF, 2015-05-12:
Setup a LaTeX template and start entering some topics for FCB documentation

adam-at-epsilon, 2015-05-18:
I suggest that we use Dia for generating UML diagrams, the we can export diagrams as .png and use them in the document.

We also need to keep the .dia files in the documentation tree so that sketches and diagrams can easily be edited in the future.

Dia is free and works both in Windows- and Linuxverse.

DanielStenis, 2015-05-19:
I've set up a LaTeX template (.tex file) in the ONE project site documents directory (/Embedded/Flight Control Board).

There are some built-in UML generators for LaTeX - could be an option as well.

adam-at-epsilon, 2015-05-19:
I'll evaluate this one a little.

http://metauml.sourceforge.net/old/index.html

DanielStenis, 2015-05-19:
Ok, the LaTeX documentation is now readded to the master branch (removed from ONE project site)

adam-at-epsilon, 2015-05-29:
the report template is now added to the 

https://github.com/afconsult-south/dragonfly doc directory under the name dragonfly-report-template.tex and I think the "master" template should live in that project.

The hyperref package has been added so that the TOC titles are clickable.