afry-south/dragonfly-fcb #47: error handling strategy

adam-at-epsilon, 2015-05-25:
We need an error handling strategy.

For instance, upon errors, we could call an error handler which switches on every red LED among the circular ones, for instance.

DanielStenis, 2015-09-10:
Is this done? @adam-at-epsilon

adam-at-epsilon, 2015-09-11:
@DanielStenis you are of course right in that different errors need different handling. Some should be logged to USB, others need corrective action (land immediately) and so forth.

I'd like to keep this issue open as a reminder.

DanielStenis, 2015-09-12:
You are right. And right now we sometimes seem to end up in the error handler. Perhaps we should print some information. For this, we should use the CDC_Transmit function and somehow bypass the mutex so we can output error data even if the RTOS/system state has failed.

adam-at-epsilon, 2015-09-12:
just an idea: in an error condition, we should repeatedly try to print every 5 seconds or so to USB so that when (or if) we attach the USB cable, FCB will immediately print a status report.