afry-south/dragonfly-fcb #64: Implement command to print active threads and status

DanielStenis, 2015-07-27:
See FreeRTOS Plus CLI command samples

vTaskList

DanielStenis, 2015-07-28:
http://www.freertos.org/rtos-run-time-stats.html

Need to provide some defines and portCONFIGURE_TIMER_FOR_RUN_TIME_STATS() function that provides a timer/counter with suitable time base.

DanielStenis, 2015-09-08:
@khayyami Is this done now then? Can we close this issue?

khayyami, 2015-09-10:
This is done, unfortunately I tagged it as #69 in my commits... sorry. To see task status use command "task-status".

khayyami, 2015-09-23:
Task status command "task-status" is implemented but it needs to be verified in order to now that the values are accurate.

adam-at-epsilon, 2015-09-27:
I think we should add the task stack high water mark functionality into the task list, this will help with task #33 .

http://www.freertos.org/uxTaskGetStackHighWaterMark.html

adam-at-epsilon, 2015-09-28:
`vTaskGetRunTimeStats` uses `sprintf`, not `snprintf` this causes a buffer overrun.