afry-south/dragonfly-fcb #76: Make use of the ARM-Cortex M4 DSP

DanielStenis, 2015-08-17:
Surely, there must be plenty of applications where we could use the Digital Signal Processor (DSP) built into the M4 core.

DanielStenis, 2015-08-19:
Check path STM32Cube_FW_F3_V1.1.0\Drivers\CMSIS\DSP_Lib

adam-at-epsilon, 2015-09-04:
There seems to be many matrix operations implemented in <code>arm_math.h</code> and even some PID controller.

DanielStenis, 2015-09-04:
We need to look at this eventually when we start using transformation matrices, calculate sensor covariance values for the Kalman Q/R values, mean calculations etc etc.

But we probably need to learn more about fixed point float (Q15, Q31). Maybe @khayyami can teach us? :)

adam-at-epsilon, 2015-09-12:
It seems that we have hardware support for conversion between `float`/`double` and fixed-point formats in the Cortex M4 processor

_It also provides conversions between fixed-point and floating-point data formats, and floating-point constant instructions_

http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.ddi0439b/BEHJADED.html

DanielStenis, 2015-09-24:
I have configured the project to use the float32_t variants of the DSP instructions. Since we have an FPU, this should work well. No need for fixed-point float?

adam-at-epsilon, 2015-09-25:
Ok, there is no point in compiling code that we're not using.

At this stage we don't have any use for fixed-point arithmetics.

DanielStenis, 2015-11-05:
It seems we have investigated this and imported the "DSP Library" into the project to handle stuff like vector and matrix operations, sin/cos functions etc.

Closing this.

closed 2015-11-05 by DanielStenis