afry-south/dragonfly-fcb #107: Sensor x y z coordinate frames

adam-at-epsilon, 2015-09-07:
Determine how x, y and z axes are oriented for gyroscope, magnetometer and accelerometer, whether they are in agreement or need rotation before use.

adam-at-epsilon, 2015-09-07:
The coordinate frame of our quadcopter fuselage is:

Positive x points forward between M1 and M2 arms.
Positive y becomes starboard.
Positive z is down.

adam-at-epsilon, 2015-09-10:
Gyro & accelerometer analysed (they have their axes according to their respective datasheets, but not aligned)

In progress.

adam-at-epsilon, 2015-09-11:
[image]

adam-at-epsilon, 2015-09-12:
The inclination of earth's magnetic field in Malmo is high, about 70 degrees.

The magnetometer shows positive values for magnetic north (cross-checked with actual compass).

http://gravmag.ou.edu/mag_earth/mag_earth.html

adam-at-epsilon, 2015-09-12:
diagram now done ... under the assumption that I got the orientation of the STM32F3 board correct with respect to the body frame of the quadcopter. Did I, @DanielStenis ?

[image]

adam-at-epsilon, 2015-09-12:
![](https://github.com/afconsult-south/dragonfly-fcb/blob/master/doc-latex/images/CoordinateFrames.png)

adam-at-epsilon, 2015-09-12:
This is hopefully clear how the sensors are orientated with respect to the quadcopter body frame.

Is the STM32F3 orientation correctly depicted (i.e. the "diode circle" points ahead?)

[image]

adam-at-epsilon, 2015-09-14:
done with commit referenced just above. Diagram of coordinate frames added to wiki page "Sensors"

closed 2015-09-14 by adam-at-epsilon

DanielStenis, 2015-09-15:
Looks right! 

Maybe we could make an enum struct that makes it possible for us to redefine the coordinate system relative to the board?