afry-south/dragonfly-fcb #112: make sense of accelerometer values

adam-at-epsilon, 2015-09-10:
Accelerometer values are too large (~ 15G) according to settings.

Find root cause and fix it.

adam-at-epsilon, 2015-09-10:
For whatever reason, using 8G as full scale value gives sensible readings.

DanielStenis, 2015-09-10:
Nice work! :) A bit strange though.

adam-at-epsilon, 2015-09-11:
Re-opened, to be continued ...

khayyami, 2015-09-11:
"There are several things on this MEMS you have to know, but which are not mentioned in the datasheet:
1.The accelerometer's output register is just 12 bits and not 16 bits, so you need to right-shift the value by 4 and multiply it with 0,001 G. Furthermore it's little-endian.
2.The magnetometer's output register is 16 bits, but big-endian. Furthermore the vector order is (X|Z|Y) not (X|Y|Z). To calculate the correct value you need to devide X and Y by 980 gauss⁻¹, while it's 1100 gauss⁻¹ for Z.
3.The temperature sensor works, but it's not calibrated. So you can use it to measure temperature change, but no absolute temperatures. It's also just 12 bits, but big-endian and you have to devide the output by 8 C⁻¹. "

http://stackoverflow.com/questions/19161872/meaning-of-lsb-unit-and-unit-lsb

adam-at-epsilon, 2015-09-11:
Fixed in lsm303dlhc.c

closed 2015-09-11 by adam-at-epsilon

DanielStenis, 2015-09-12:
@adam-at-epsilon Perhaps we should go back to using 2G resolution? Maybe you already did this?

DanielStenis, 2015-09-12:
"The accelerometer and magnetometer both output readings in a 16-bit format (obtained by combining the values in two 8-bit registers for each axis), but they contain a maximum of 12 bits of precision. For the accelerometer, at least the lowest 4 bits of the output values are always 0, and for the magnetometer, the highest 4 bits of the output values are always 0."

DanielStenis, 2015-09-12:
Inside the accelerometer is a proof mass. It is a little piece of metal that is suspended with springs. The accelerometer measures the distance this proof mass is displaced.
Ignoring gravity, if you were to move the accelerometer upwards, the proof mass will be displaced towards the bottom of the accelerometer (because its inertia wants to keep it where it is). The accelerometer is programmed to interpret this as an acceleration in the upward direction.
When the accelerometer is resting on the table, gravity is pulling the proof mass towards the bottom of the accelerometer, similar to above, so it interprets that as an acceleration in the upward direction.

DanielStenis, 2015-09-12:
@adam-at-epsilon @khayyami Found some interesting stuff about the accelerometer/magnetometer (see above posts). Explanation on positive Z-axis values due to gravity.

Also, both sensors are 12-bit, just aligned differntly.

@adam-at-epsilon Are we using high-res mode on accelerometer?

adam-at-epsilon, 2015-09-12:
@DanielStenis good finds on the info, it will come in very handy for sorting out the magnetometer.

Yes, we're setting LSM303DLHC_HR_ENABLE for CTRL_REG4. We're using 2G mode as well.

In 8G full scale mode, we have a resolution of 4 milli-G, but with a G being 9.82m/s2, that then translates into 4cm/s2 acceleration, not good enough if we want to fly through doors.