afry-south/dragonfly-fcb #116: Calculate attitude with magnetometer

DanielStenis, 2015-09-11:
Since the earh magnetic field is static (unlike acceleration) and therefore unaffected by the quadcopter movement, it would be nice if we could use this to sense attitude (and use it as correction for roll/pitch/yaw Kalman filters).

Figure out a way to calculate this. Calculate for the magnetic field vector declination and inclination for Malmö, Sweden.

khayyami, 2015-10-02:
Implemented GetMagYawAngle in rotation_transformation.c

Algorithm:
Y = magX * arm_sin_f32(roll) * arm_sin_f32(pitch) + magY * arm_cos_f32(roll) - magZ * arm_sin_f32(roll) * arm_cos_f32(pitch);
X = magX * arm_cos_f32(pitch) + magZ * arm_sin_f32(pitch);
yawAngle = atan2(Y,X);

DanielStenis, 2015-10-04:
It seems this may be alot tricker than expected.

However, the rotation can be represented by rotation by an angle around a perpendicual vector in 3D space.

Using Rodrigues rotation formula to retrieve the DCM, roll pitch and yaw can be extracted!

http://math.stackexchange.com/questions/285382/solve-for-angles-in-a-matrix/285463#285463

DanielStenis, 2015-10-07:
This still needs testing. Try printing values and see how good they are.

Perhaps this will not be used, or maybe only for yaw.

Or perhaps one could fuse data from magnetometer attitude with accelerometer.

DanielStenis, 2015-10-23:
Putting this issue on hold for now. It seems quite difficult to use the magnetometer to get all of roll, pitch and yaw although it should work in theory. It work OK with roll and pitch, but for yaw it didn't work at all and I don't know why.

However, it is unclear if this would be a feasible method to use going forward if the accuracy of the magnetometer was greatly affected by the motors.

The general idea was:
- Get the axis/angle rotation between inertial and body/sensor magnetic vectors
- Convert this to Euler Angles (ZYX also known as Tait-Bryan Angles)

Along these lines and using axis-angle/quaternion relationships with Euler angles.
http://math.stackexchange.com/questions/285382/solve-for-angles-in-a-matrix/285463#285463

Not sure if it was a problem of the axis/angle rotation was not unique?

DanielStenis, 2015-11-05:
It might be an idea to post on math.stackexchange if/when this is revisited

adam-at-epsilon, 2015-11-06:
or the ÅF forums?