afry-south/dragonfly-fcb #117: Create quadcopter simulator in Scilab

DanielStenis, 2015-09-11:
Scilab is a free numerical mathematics platform, similar to Matlab. It features the XCos simulation block-building tool useful for modeling dynamical systems (like Simulink).

Model the 6-DoF quadcopter model
* 6-DoF body-frame dynamic equations
* Rotation matrix to world-frame
* Model PID controllers with output representing forces and moments

adam-at-epsilon, 2015-11-05:
@DanielStenis should we repurpose this to include use python/numpy instead? We could also perhaps ask for the simulator code from the AUTONAVx code, I saw something about that in the course forum.

DanielStenis, 2015-11-05:
I guess we could leave it open for now and whoever gets down to doing it can decide.

It could be a valuable exercise to construct and own one to get acquainted with the dynamical system equations and transformations between coordinate systems.

I found a Python package for dynamical system simulation, PyDSTool: https://wiki.python.org/moin/PyDSTool