afry-south/dragonfly-fcb #120: create __FCB_RELEASE/DEBUG/TEST__ macro #defines

adam-at-epsilon, 2015-09-12:
Configure preprocessor macro defines named for configurations

Debug: `#define __FCB_DEBUG__`
Release: `#define  __FCB_RELEASE__`
Test: `#define __FCB_TEST__`

This will allow us to exclude test/debug code from the release, or have special settings for the release version.

adam-at-epsilon, 2015-10-09:
We now have __FCB_DEBUG__ and __FCB_RELEASE__, but __FCB_TEST__ will have to wait until whenever we have a `Test` release.

adam-at-epsilon, 2015-10-09:
This will help us differentiate for instance error handling and printing to console behaviour between test/debug/release phases.

DanielStenis, 2015-10-14:
I agree. Perhaps we should make a few more build configurations with different project define setups?

Maybe we should consider a #define that disables any writing to the motors for safe testing?

And perhaps another one that simply runs some unit tests?

adam-at-epsilon, 2015-10-19:
All good suggestions. One build configuration for unit test and perhaps another for calibration?

Perhaps the motors should be disabled by default in the and we have to add a define to enable them?

DanielStenis, 2015-10-19:
Probably a good idea, yes. One could always update the motor values, but not outputting them on the actual pins unless we define MOTOR_OUTPUT_ENABLED