afry-south/dragonfly-fcb #122: Lab experiment: Motor rpm time const.

DanielStenis, 2015-09-14:
In order to tune our control algorithm as well as possible, we need to investigate how fast the motors (with propellers) respond to updated control commands.

* Lab/experiment setup with some sort of rpm-sensor. Perhaps use a simple photoresistor on one side of the propeller and a light source on the other. Hook up the resistor to an oscilliscope and note 2 pulses for each round (since we have two propeller blades).
* Calculate approxmiate motor time constant for some different commands
* Note that the motor has a soft start feature when starting the motors from zero, so make sure this is not in use when conducting experiments.
* Do some plots on motor control value and rpm change (both for a step response from one motor control value to another and an overall stationary control value<->rpm map)

adam-at-epsilon, 2015-09-16:
Apparently there are reasonably accurate tachometers available for Android smartphones.

http://www.svensktmodellflyg.se/forum1.asp?viewmode=1&msgid=280601

adam-at-epsilon, 2015-09-16:
Hobby tachometers are priced at 199SEK here.

http://www.mbs-rcmodels.se/mat-och-testinstrument-2/optisk-varvrakare/

adam-at-epsilon, 2015-11-02:
Our .pdf document says the maximum thrust is 1230g in Table 9.1, that is equal to 12.3 N

Here's an available dynamometer for 20N for about 100SEK:
http://www.heraco.se/groups.aspx?arID=3&gID=1777&filter=M

adam-at-epsilon, 2015-11-16:
the dynamometer has arrived and is stored in the Dragonfly storage box.