afry-south/dragonfly-fcb #124: Use separate LaTeX documents for theory and implementation?

DanielStenis, 2015-09-16:
Perhaps it's a good idea to have separate documents on theory and code design and implementation.

Theory:
- Flight dynamics (coordinate systems, system equations etc.)
- Flight control
- Estimation theory
- Motor theory
- Sensor theory
- Experiments

Implemenation:
- Code design
- UML diagrams
- Real-time diagrams on running tasks etc.

adam-at-epsilon, 2015-09-16:
The only difficulty I can foresee with having two documents is the usability of the documentation on the part of the reader.

If we have one document, we can click cross-references back and forth between theory and implementation, if we have two different documents this back-and-forth clicking is not possible because links in PDF cannot open other PDF documents as far as I'm aware.

DanielStenis, 2015-09-16:
I guess.

But regardless it would generally be nice to separate these theory and implementation. Perhaps in chapters then.

adam-at-epsilon, 2015-09-16:
Yes I agree they should be separate chapters. If the documents get too big & unwieldy to edit & maintain perhaps we should split them into separates.

DanielStenis, 2015-09-17:
Then maybe this is a non-issue at this moment in time. Closing.

closed 2015-09-17 by DanielStenis