afry-south/dragonfly-fcb #132: Kalman filter division by zero possible

DanielStenis, 2015-10-07:
In the correction step, there is a possibility that a division by zero occurs - ruining the Kalman filter output for the length of the program operation.

We need to protect us from this somehow and figure out a way to minimize the chance we get close to this.

@khayyami has some thought and information on this.

DanielStenis, 2015-10-24:
Some of the legwork on this issue has been done in issue #119.

I have come across information regarding numerical instability in the standard Kalman filter design, possibly caused by float rounding. This could result in the covariance matrix P becoming non-semidefinite which just ain't good.

This problem is mentioned on Wikipedia https://en.m.wikipedia.org/wiki/Kalman_filter along with modifications to address this problem.

I feel confident we're having these sort of problems and I think it's more likely to happen when board is standing still.

We should print a whole bunch of filter stuff to verify this. Especially when NaN output detected.

Then read up on potential fixes and implement them.

adam-at-epsilon, 2015-10-25:
A few days ago I started to refine our printouts using a CALIBRATION_SERIALIZATION setting but I haven't merged this to `master` yet.

Yes, the P11 and P22 elements keep deminishing.

adam-at-epsilon, 2015-10-25:
The Kalman Gain is going to 1, which implies our measurements are given no weight.

DanielStenis, 2015-10-28:
An error in the Kalman correction update of covariance matrix was detected and fixed. This now works. Update documentation with a fix to the calculation error (see code in issue119 branch)

adam-at-epsilon, 2015-10-28:
we repurpose this issue to fix the Not-a-Number and the division by zero :8ball:

DanielStenis, 2015-10-28:
We should close this issue after updating documentation on Kalman filter and pushing new code to master. Feeling quite happy with this now! :+1:

adam-at-epsilon, 2015-10-29:
This commit should cover a lot of it, even though it had the wrong tag (#119): 14d0f7e

adam-at-epsilon, 2015-10-30:
Ran another overnight test on aforementioned commit and the filter seems stable after an uptime of 15 hours.

So stability of the KF filter seems to be fixed. I leave the honour of closing this ticket to you, @DanielStenis :+1:

DanielStenis, 2015-10-30:
Doing it!

closed 2015-10-30 by DanielStenis